sinara-hw/HV_AMP_8CH #1: initial spec

gkasprow, 2020-12-02:
We need this amplifier for AEGIS Ion trap control.
At lest 32 channels are needed so we will use Fastino and add 4 amplifier cards on its right side connected using flat cables.
additional EEMs will be used to directly control the opto-MOS relays on the amplifier outputs
I plan to use slighty modified discrete amplifiers used in Stabilizer Piezo Driver design. After replacing FETs with ones that work up to 450V I should be able to work in +/-200V range

gkasprow, 2020-12-02:
initial components placement
[image]

jordens, 2020-12-09:
Nice. A lot of the macroscopic trap people will like that. Is there a noise/bandwidth spec already?

gkasprow, 2020-12-16:
they say "the lower the better" :D

gkasprow, 2020-12-20:
Routing is done
[image]

dtcallcock, 2020-12-21:
What's the tempco likely to be?  Would be good if this amp doesn't significantly degrade Fastino in this regard.  Will the R10/R17 divider be the limit?

gkasprow, 2021-05-10:
I used 10ppm resistors so they seems to be the limit