sinara-hw/HV_AMP_8CH #2: debugging

gkasprow, 2021-01-14:
The boards arrived
[image]
[image]

jordens, 2021-01-15:
Very nice!

gkasprow, 2021-01-15:
The HV discharge circuit causes HV converter overload during startup -> we have a 1.3A startup current which decays to 0.3A after a few seconds.
The circuit works properly and quickly discharges capacitors, but the 1k resistors are sparking, which means they must be replaced with HV version

gkasprow, 2021-01-15:
R81 is too high - it does not let enough current to flow to T10 emitter. It requires at least 0.6mA
Removal of R84 solves the issue.

gkasprow, 2021-01-15:
I suggest adding the 8V2 Zener diode in series with R81

gkasprow, 2021-01-15:
- [ ] R84 -> 47k
- [ ] R81 -> D8V2 + 3k3
- [ ] R75,R83 -> 10k+10k 1206. Existing 1K get damaged during discharge. After replacement by 1+1k they get damaged, too

gkasprow, 2021-01-15:
After the changes below, the HV disappears within 3s after 12V is disconnected so it is acceptable.

gkasprow, 2021-01-15:
the HV supplies produce +/- 232V instead of ~+/-205V

gkasprow, 2021-01-15:
It works nicely at +/-12V, but when HV is enabled, the output FETs get hot and the viltage is limited to +/-60V - the amplifier is oscillating at 4...5MHz causing obvious current overload

mrdorababu, 2021-01-19:
HI.... 
High voltage dc-dc converter both sides are having the same ground. If we are using the same ground plane, these grounds may be shorted.

gkasprow, 2021-04-29:
The oscillations are caused by over-current protection.
The transistors were designed to limit the upper and lower rail current by lowering output MOSFET VGS  voltage. Let's assume the upper MOSFET current is exceeded. The BJT opens and tries to lower the upper MOSFET VGS, but the gate signal is propagated to the lower MOSFET gate causing it to increase the current. The lower side overcurrent  BJT starts conducting, trying to lower the lower side MOSFET voltage, which is then transferred to the upper side MOSFET, and the situation repeats.
The remedy is simple - disable the protection with BJTs. and just lower the protection Zener diode voltage.
Another situation is with the biased current of output MOSFETs. It increases with temperature. The MOSFET temperature coefficient is positive for low currents and negative for high currents. That's why in audio power amps such compensation is often not needed, unlike the case of BJTs.

NFET:
[image]

PFET:
[image]

gkasprow, 2021-04-29:
So we have to compensate the Vgg.
We have limited MOSFET data, at 100mA the Vgs changes by 0.5V between 25C and 100C. It gives roughly  0.5V/70C = 7.1mV/K. At lower currents, the coefficient will likely be higher.
We can use 4 Si diodes connected in series to get 8mV/K  and add such slope in series with the upper and lower FET biasing resistors. They are controlled by a constant current source driven by an opamp, so that should provide some temperature compensation.

gkasprow, 2021-04-30:
We also need to add TVS that protects other opamps in case one channel fails.
In case of failure, it can happen that too high voltage appears at the opamp output and can be distributed via supply network to other channels.

gkasprow, 2021-05-01:
The amplifiers work very nicely, but with a 1MHz sinewave, they consume more than 25mA of current (10W of power). Moreover, they are very fragile to supply ripples due to the AC connection between the gate and opamp output. The PSRR is close to 0dB or even negative. Multi-channel construction must have a dedicated LDOs for both rails. Another issue is that the HV power on ramp must be slow, otherwise, both FETs get enabled. It seems this architecture is not suitable for that application and we should substitute these discrete amps with more typical topology.

gkasprow, 2021-05-01:
the temperature compensation implemented as four silicon diodes in series (BAV199S) actively reduce the current while heating. So the amplifier is thermally overcompensated which gives negative temperature feedback.

gkasprow, 2021-05-02:
Even mV ripples at the output of the power supply LC filter get amplified to tens of V. The effect is interesting because at the startup the amplifier output is still DC. The small bias heats the FETs which have a positive temperature coefficient, causing the current to rise to increase the power drawn from the supply. The thermal compensation diodes are still cold. But once the HV supply-demand increases, the ripples get higher. Since the PSRR is poor, they are amplified and at the output, I observe tens of V at 200kHz, which is the switching frequency of the converter. This causes further heating of the output stage, increasing the ripples. After a few seconds their amplitude is over 100V and gets stabilized because the thermal compensation diode got heated and decreased the bias current.
The amplifier construction is interesting because:
-  it has over 2MHz bandwidth. 
-  it works well from the stabilized linear power supply. 
- when working at high frequency, it draws a lot of power due to the cross-conduction effect. The transfer characteristics of FETs are far from being linear, and when applying the sinewave, there are some voltage regions where the cross-conduction current is high. I observed nearly 30mA drawn from DC power at 1MHz without load. It's a lot.
- it could be a good solution for pulse applications that require high bandwidth and very low Iq. The amps must have a dedicated LDO with good PSRR.

gkasprow, 2021-05-02:
For this particular application, we will use standard APEX high-voltage amplifiers. To get +/-200V output, we have to increase the voltage to +/- 220V. We can use existing power converters (100V version) but we have to connect them in series.

closed 2021-05-10 by gkasprow