sinara-hw/HV_AMP_8CH #4: v1.1 arrived

gkasprow, 2021-06-05:
[image]

jordens, 2021-06-05:
Is that with apexes? Could you upload the release data and PDF for that? I'd like to follow this development!

gkasprow, 2021-06-05:
yep. Done https://github.com/sinara-hw/HV_AMP_8CH/releases/tag/v1.1

gkasprow, 2021-06-21:
Full power BW
[image]

at 50kHz it's able to output 260Vpp
[image]

at 100kHz it's able to output 124Vpp
[image]

at 250kHz - 42Vpp
[image]

at 500kHz - 26Vpp, limited by SR
[image]

at 1MHz - 13Vpp limited by SR
[image]

square wave at 33kHz. We can clearly see SR of 30V/us as in the datasheet. It means that the input amp didn't break the spec. We need input amp due to HV amp GPB.
[image]

gkasprow, 2021-06-21:
Thermal protection works. After heating the temp sensor the HV and 24V rails get disabled.