sinara-hw/HV_AMP_8CH #7: stability and noise performance

jordens, 2021-11-29:
@gkasprow Do you or the AEGIS people already have any data on the stability, the temperature coefficient and the noise?

gkasprow, 2021-11-29:
that's a good question to @kaolpr

kaolpr, 2021-11-29:
AFAIK not yet. At TS we did only basic tests and AEgIS team is currently solving some problems with their setup. Don't think they managed to do any test before that. However, if any information comes up I'll inform you.

andrewvh4, 2022-02-21:
Are there any updates on these noise and stability performance measurements?

gkasprow, 2022-02-21:
@andrewvh4 CERN discovered that the amplifier is unstable with a long cable. I did several measurements with a modified compensation loop.
Here are the results. The screenshots have self-explanatory file names. The first case was with added 2.4pF capacitor. The second case was with added 8pF capacitor.
https://www.dropbox.com/sh/22puqkn01m83u8a/AADVKAsE6MOcG1ZhXLNaRnrHa?dl=0

gkasprow, 2023-04-03:
finally, 100pF 1808 capacitor will be used.