sinara-hw/HV_AMP_8CH #10: Low power PA97 variant

jordens, 2022-07-13:
For low slew rates and low quiescent power applications one could make a variant for the PA97. It's almost pin compatible and doesn't need cooling.

gkasprow, 2022-07-26:
good idea. 
Btw, is not really related, but I made a list of Apex HV amplifiers.
[image]
[Apex_amps.xlsx](https://github.com/sinara-hw/HV_AMP_8CH/files/9191099/Apex_amps.xlsx)

jordens, 2023-04-25:
@gkasprow could you make a variant of v1.2 for PA97?

gkasprow, 2023-04-25:
It looks the PA97 can be installed without any modifications. Output is pin8 instead of 7, but we have their 3R6 resistor anyway, which won't make any difference because it's inside the feedback loop.

jordens, 2023-04-25:
What about BOM changes (no heatsink etc.)?

gkasprow, 2023-04-25:
sure, this need to be done but does not block the PCB review.

kaolpr, 2023-05-11:
> It looks the PA97 can be installed without any modifications. Output is pin8 instead of 7, but we have their 3R6 resistor anyway, which won't make any difference because it's inside the feedback loop.

It's not true. PA97 has two package styles: PC and DR (DR is NRND) whereas in HVAMP there is PA95 in DQ package used.
Those packages are not pin compatible.

jordens, 2023-05-11:
I don't see the problem.

kaolpr, 2023-05-11:
My mistake. It's fine. Although, we could make an assembly variant anyway.

gkasprow, 2023-05-11:
I will once I get CERN symbols

jordens, 2025-01-15:
Works fine. Closing.

closed 2025-01-15 by jordens